SmartEVSE/SmartEVSE-2 #2: Reading Modbus interface EVSE

rvdgaag, 2021-03-06:
Hi,

I'm trying to read registers via Modbus RTU from the EVSE with iobroker and a USB-modbus converter.
All works well reading the SDM630 I connected as well but I don't get any values for the EVSE.

As an example I try to read the register 0x00C0 Configuration value 0 socket, value 1 fixed.
I've set in the menu of the EVSE the configuration to fixed cable, still reading this register comes with value 0.

Probably the register setting is wrong in iobroker, is it a input register, a coil or a holding register ?

Since these registers are in HEX format I need to convert them, 192 right ?

Maybe the issue is that the EVSE is a Master and probably the iobroker (raspi 4) as well ?

Thanks for your feedback!

bobosch, 2021-03-06:
Hi,

when you use load balancing mode it's an issue with two masters. SmartEVSE sends multiple requests every two seconds - you have to try multiple times to get a valid response.

Have you tried communication when load balancing and smart/solar mode is disabled?

rvdgaag, 2021-03-06:
Thanks for the feedback,

yes I did not much better results. I was hoping to change parameters from remote, any ideas ?

bobosch, 2021-03-07:
Maybe wrong byte order. Set it on the SmartEVSE to fixed configuration and read value again.
Yes, it is a input register (FC=04).
Yes, addresses are in HEX format (0x indicates that)
Also correct settings? 9600bps 8N1 (8bit, no parity bit, one stop bit)
Do you have a modbus electric meter? Does your adapter work with this?

TonKro, 2021-04-11:
I am trying to get the data using Node-Red using ModBus module. Somehow I could not get it working.

With serial port I see the data Reqeust and Answer:

Request:
[10,4,0,0,0,20,241,126]

Answer data:
[10,4,40,0,21,0,3,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63,234,121,245,63,180,28,153,62,123,167,51,113,98]

I am new with modbus, can someone explain the data? Or how can I translate this?

Using modbus I do need to have some data to read out:
[image]

What should be the Unit-ID / Address / Quantity?

bobosch, 2021-04-12:
No, all values from SmartEVSE are 16bit unsigned integer -> 20 registers = 40 bytes
Maybe first try reading a specific register. Use this information when you using 2.20 or older
https://github.com/SmartEVSE/SmartEVSE-2/tree/b8b6a1a2def22a2bfc3cc30be2f2ef639e2d4389#register-for-external-devices

TonKro, 2021-04-15:
That is also what I thought, but it was not working.
Apperently it is 32 bit floating:

Data of registers 0,21 0,3, 0,0, 0,0, 0,0, 0,0, 0,0, 0,0, 0,0, 0,0, 0,0, 0,0, 0,0, 0,0, **63,234, 121,245, 63,180, 28,153, 62,123, 167,51,**

Phase 1: 63,234,121,245 = 1,83
Phase 2: 63,180,28,153   = 1,41
Phase 3: 62,123, 167,51 = 0,25

I did (for Node-Red) use:

var V1 = new Buffer ( [msg.payload[31], msg.payload[32], msg.payload[33], msg.payload[34] ] ); 
var V2 = new Buffer ( [msg.payload[35], msg.payload[36], msg.payload[37], msg.payload[38] ] );
var V3 = new Buffer ( [msg.payload[39], msg.payload[40], msg.payload[41], msg.payload[42] ] );

var msg1 = {payload:V1};
var msg2 = {payload:V2};
var msg3 = {payload:V3};
 
return [msg1,msg2,msg3];

And then convert msg1/2/3 these to floating.

However I cannot find this information back in the documentation? Or do I read the wrong documents?

bobosch, 2021-04-15:
Sorry, I mixed it up. SmartEVSE are 16bit integers, Sensorbox has 32bit float.
Yes, Sensorbox code and documentation is not published yet.
https://github.com/SmartEVSE

TonKro, 2021-04-15:
Can you explain more detailed? What is the modbus server and what is the client? I assume the SmartEVSE (master) does do the request ([10,4,0,0,0,20,241,126] CRC), and the Sensorbox does reply and writes into the SmartEVSE with eg data:
[10,4,40,0,21,0,3,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63,234,121,245,63,180,28,153,62,123,167,51,113,98]?
Is there a somewhere an overview?

mstegen, 2021-04-16:
The master (SmartEVSE) requests the data from the Sensorbox , or other modbus kWh meter.
The Sensorbox or kWh meter, replies with the requested data.

The measurement data from the Sensorbox 1.5/2.0 is in float32 (one float32 for each phase)

 ```
 Register  Register  
  Address   length (16 bits)
  0x0000      1       Sensorbox version 1.5           = 0x000F
  0x0001      1       DSMR Version(MSB), CT's (LSB)   = 0x0003                 
  0x0002      2       Volts L1 (32 bit floating point), Smartmeter P1 data. Unused on Sensorbox 1.5
  0x0004      2       Volts L2 (32 bit floating point), Smartmeter P1 data. Unused on Sensorbox 1.5
  0x0006      2       Volts L3 (32 bit floating point), Smartmeter P1 data. Unused on Sensorbox 1.5
  0x0008      2       Amps L1 (32 bit floating point), Smartmeter P1 data. Unused on Sensorbox 1.5
  0x000A      2       Amps L2 (32 bit floating point), Smartmeter P1 data. Unused on Sensorbox 1.5
  0x000C      2       Amps L3 (32 bit floating point), Smartmeter P1 data. Unused on Sensorbox 1.5
  0x000E      2       Amps L1 (32 bit floating point), CT imput 1
  0x0010      2       Amps L2 (32 bit floating point), CT input 2
  0x0012      2       Amps L3 (32 bit floating point), CT input 3
```

TonKro, 2021-04-16:
Clear overview thanks that makes much sense.

In my example for register 0 and 1:
0,21 0,3, (decimal)
Register 0x0000 0x00 0x15    Which sensorbox version is this?  
Register 0x0001 0x00 0x03

mstegen, 2021-04-16:
That's version 2, also indicates grid type and grid rotation:
`
0x0014 (2 lsb mirror the 3/4 Wire and Rotation configuration data)  
0x0015 = 4 wire, CCW rotation  
0x0016 = 3 wire, CW rotation  
0x0017 = 3 wire, CCW rotation
`

TonKro, 2021-04-16:
Thanks a lot for your help.