SmartEVSE/SmartEVSE-2 #3: SPM1-AC Temco current sensor compatibilty ?

NicolasFOVET, 2021-03-21:
Hi :)
Sorry if it's the wrong place to ask for this, but I'm really interested in building a charge station with the Smart EVSE and I was wondering if it could be compatible with the RS458/Modbus SPM1-AC Temco meter. The datasheet is here : https://downloads.temcocontrols.com/file/single-phase-power-meterwith-internal-ct/brochure_pdf/latest

Thanks for your great work ! :)
Nicolas

bobosch, 2021-03-21:
Hi!

No, only 3-phase electric meter are currently supported.

NicolasFOVET, 2021-03-22:
Ok, understood.
But, Smart ESVE is compatible with a single phase installation, right ? If I go with the sensorbox v1.5, do I have to connect all 3 clamp sensors ?

mstegen, 2021-03-23:
Yes, the SmartEVSE will work on one phase installations too.
If you use the Sensorbox 1.5 or 2 you can connect up to 3 CT's. If you connect only one, that will work as well.