SmartEVSE/SmartEVSE-2 #4: Eastron SDM630 config

wbitholedotbe, 2021-03-25:
I bought the smart evse with sensor box. I also bought the eastron SDM630 v2. How to configure the eastron so it will show the consumption of the charger on the display?

mstegen, 2021-03-26:
You need to enter the setup menu of the Eastron, and set the modbus address (Set Addr) on the SDM630 to a number >10 , for example 101.
Set the baudrate ( Set Baud) to 9.6k
Set the parity (Set Pari) to NONE
Set the stop bit (Set Stop) to 1

Then on the SmartEVSE set the "EV METER" option to Eastron.
set the modbus address "EV ADDR" of the meter to the same address as you used on the SDM630 (so here 101)

Connect the RS485 wires to the A and B terminals of the meter. 
If the connection is working, you will see a telephone symbol on the display of the meter.

wbitholedotbe, 2021-03-26:
Will try it this weekend. Thank you.

fluppie, 2021-08-06:
I have a monophase setup. Does this also work with the Eastron SDM220?

fluppie, 2021-08-11:
Can it be as simple by changing the custom unit to the 1P modbus register?
The SDM630 uses 0x34 and the SDM120/220/230 uses 0x0C.

`{"Eastron1F", ENDIANESS_HBF_HWF, 4, MB_DATATYPE_FLOAT32,    0x0, 0,    0x6, 0,   0x0C, 0,  0x156, 0}  // Last entry! // use 0x0C for single phase SDM120/220/230 instead of 0x34`

Should one create a case EM_CUSTOM?

bobosch, 2021-08-12:
It depends how you want to use the single phase electric meter.
I don't have this electric meter so I could only speculate...

For current measurement (smart or solar charging):
You have to use EM_EASTRON, as there is a special handling of the current measurement (case EM_EASTRON...)
I think it won't work with the SDM220, because there is no power measurement. Maybe with the SDM230, it depends on how the electric meter response the the modbus request on register 0x06 with 6 requested values (3x phase current, 3x phase power). When it response with the 1st phase current and power information, and the other values 0, it should work out of the box.

For power and energy measurment (ev meter):
You can use EM_CUSTOM with register 0x0C for power measurement (as you wrote) and 0x156 for energy measurement and it should work

Feedback is welcome!

fluppie, 2021-08-13:
I would think it can work, have you seen the registers overview from the Reaper7 repository?
Starting from line 100: https://github.com/reaper7/SDM_Energy_Meter/blob/8bda969f312cfafa5a26506da8ac9b42d4fb38ca/SDM.h#L100
[image]

So 0x06 gives my current and 0x0C my power, the same address as for Phase1_Current and Phase1_Power. The registers of Phase2 and 3 are not used by the SDM220, so I don't know what the plugin does if that value is not available. Like you wrote, if it's zero, it's probably going to work just fine.
I would also think when the system needs the total system power 0x0C would be right instead of 0x34 in a 3 Phase system.

Maybe we can create Eastron1F and Eastron3F in EVSE.h ? I think people with mono phase systems wouldn't mind :-).

mstegen, 2021-08-16:
I've got confirmation from a customer that the Eastron SDM230 modbus works fine. If the SDM220 uses the same registers then it might also work.

fluppie, 2021-08-16:
It is for most addresses. 

`{"Eastron",   ENDIANESS_HBF_HWF, 4, MB_DATATYPE_FLOAT32,    0x0, 0,    0x6, 0,   0x34, 0,  0x156, 0}, // Eastron SDM630 (V / A / W / kWh) max read count 80`

0x0 for Voltage is the same for SDM630/SDM230/SDM220
0x6 for current is the same for SDM630/SDM230/SDM220
0x34 is only used by SDM630 for SDM_TOTAL_SYSTEM_POWER, NOT supported by SDM230/SDM220. 1Phase meters use 0x0C as address for power.
0x156 for SDM_TOTAL_ACTIVE_ENERGY is the same for SDM630/SDM230/SDM220

Or is the 0x34 address not important? Maybe the load balancing uses current instead of power?

Can adding a line of code work and create an option for 1Phase and 3Phase meters?
`{"Eastron1F",   ENDIANESS_HBF_HWF, 4, MB_DATATYPE_FLOAT32,    0x0, 0,    0x6, 0,   0x0C, 0,  0x156, 0}, // Eastron SDM630 (V / A / W / kWh) max read count 80`
`{"Eastron3F",   ENDIANESS_HBF_HWF, 4, MB_DATATYPE_FLOAT32,    0x0, 0,    0x6, 0,   0x34, 0,  0x156, 0}, // Eastron SDM630 (V / A / W / kWh) max read count 80`