SmartEVSE/SmartEVSE-2 #12: How to determine the Charging States from Terminal.

qasimali-byte, 2021-05-10:
hi, I am reading the EVSE terminal using python3. I want to read the states (Connected, Charging, Disconnected) which are not available in the terminal. But when I go through the EVSE C code I figure out (State A, State B, State C) are available in the code. but these settings are not available on the terminal. Could you please help me out with the issue?
@wido @bobosch @Links2004 @mstegen @maximevince

qasimali-byte, 2021-05-12:
@bobosch thanks for your comment. could you please answer me that can i connect the Following Chint Smart Energy meter with Smart EVSE over the modbus and show the Chint meter values on the terminal ?
[image]

qasimali-byte, 2021-05-12:
> Currently you can read the charge state only over Modbus (RS485),
> https://github.com/SmartEVSE/SmartEVSE-2/tree/b8b6a1a2def22a2bfc3cc30be2f2ef639e2d4389#register-for-external-devices

so as we already reading the terminal through rx,tx availbale on the Smart EVSE . so you are saying if  i want to read the states values then we should communicate with Smart EVSE over the RS-485 and then read the States Values ?

mstegen, 2021-05-12:
If you are using the terminal, you can also send the serial command STATE? 
The SmartEVSE will answer with:

EVSE0:B(16.0A),EVSE1:C(12.0A),EVSE2:A(0.0A),EVSE3:A(0.0A),EVSE4:A(0.0A),EVSE5:A(0.0A),EVSE6:A(0.0A),EVSE7:A(0.0A)

qasimali-byte, 2021-05-12:
> If you are using the terminal, you can also send the serial command STATE?
> The SmartEVSE will answer with:
> 
> EVSE0:B(16.0A),EVSE1:C(12.0A),EVSE2:A(0.0A),EVSE3:A(0.0A),EVSE4:A(0.0A),EVSE5:A(0.0A),EVSE6:A(0.0A),EVSE7:A(0.0A)

Thanks, @mstegen for your comment. as my question was I want to determine the states (Connected, Charging, Disconnected) I hope you understood my question. but as you answered that if I send the serial command State over the terminal. the answer of the Smart EVSE will be EVSE0:B(16.0A),EVSE1:C(12.0A),EVSE2:A(0.0A),EVSE3:A(0.0A),EVSE4:A(0.0A),EVSE5:A(0.0A),EVSE6:A(0.0A),EVSE7:A(0.0A) . but how can I extract the state from this line that Currently what is the state?

mstegen, 2021-05-12:
If it's state A, then it's Disconnected. In the above example EVSE2-7 are Disconnected.
State B is Connected (only EVSE0 , the Master in this case)
State C is Charging (only EVSE1 , it's charging at 12.0A at the moment)

qasimali-byte, 2021-05-12:
> If it's state A, then it's Disconnected. In the above example EVSE2-7 are Disconnected.
> State B is Connected (only EVSE0 , the Master in this case)
> State C is Charging (only EVSE1 , it's charging at 12.0A at the moment)

okay. so you are talking about the **LOADBL**  when we enable this we have Master, Node 1----Node 7. so we can determine the states accordingly. am I right?

qasimali-byte, 2021-05-12:
> If it's state A, then it's Disconnected. In the above example, EVSE2-7 is Disconnected.
> State B is Connected (only EVSE0, the Master in this case)
> State C is Charging (the only EVSE1, it's charging at 12.0A at the moment)
As we give only those commands to the Terminal which are available on the terminal. **STATE** command is not available on the terminal.

mstegen, 2021-05-12:
You should use STATE**?**
Yes It will display all states for all Nodes including the Master. 
The STATE? command only gives usable results on the Master.

qasimali-byte, 2021-05-24:
> If you are using the terminal, you can also send the serial command STATE?
> The SmartEVSE will answer with:
> 
> EVSE0:B(16.0A),EVSE1:C(12.0A),EVSE2:A(0.0A),EVSE3:A(0.0A),EVSE4:A(0.0A),EVSE5:A(0.0A),EVSE6:A(0.0A),EVSE7:A(0.0A)

hi! @mstegen  i run STATE? commond in Master and get the EVSE0:B(16.0A),EVSE1:C(12.0A),EVSE2:A(0.0A),EVSE3:A(0.0A),EVSE4:A(0.0A),EVSE5:A(0.0A),EVSE6:A(0.0A),EVSE7:A(0.0A) could you please tell me in the above result which is my EVSE . as There are two EVSE0:B(16.0A),EVSE1:C(12.0A) displaying how can i recognize which is my EVSE ?

mstegen, 2021-05-24:
The Master is EVSE0

qasimali-byte, 2021-05-24:
Ok, thanks. But what about EVSE1 who is also displaying some current. Is it
that way

EVSE 0--- Master
EVSE 1--- Node1
.....
EVSE 7----Node 7

qasimali-byte, 2021-05-25:
> The Master is EVSE0

Ok, thanks to @mstegen . But what about EVSE1 who is also displaying some current. Is it
that way

EVSE 0 --->Master
EVSE 1 --->Node1
.
.
EVSE 7--->Node 7

mstegen, 2021-05-25:
Yes that's correct.

qasimali-byte, 2021-05-25:
okay. Thanks, It's means when we connect Something with Node1---Node7 they will also display their status and Currents.?

mstegen, 2021-05-25:
Yes, for each connected controller the current STATE and max charge current is displayed.

qasimali-byte, 2021-05-25:
Ok, thanks for the clarification.

qasimali-byte, 2021-06-02:
hi , @mstegen  i am facing the issue regarding getting the States . as there is no Curennt Showing in EVSE Screen as well as on the STATE? Commond . can you please help me out where i am doing mistake .
[image]
[image]

qasimali-byte, 2021-06-02:
hi, i am facing the issue regarding getting the States . as there is no
Current Showing on EVSE Screen as well as on the STATE? Command . can you
please help me out where I am doing a mistake

mstegen, 2021-06-02:
Somehow the sensorbox is not measuring any current. Are you using a sensorbox or are you emulating it in software?
If you can send a picture of your complete setup, that might help.

qasimali-byte, 2021-06-02:
This is the picture of the circuit which I am using. 2nd as you mentioned
that am I emulating the current from software. Could please how can I
emulate the from the software .

mstegen, 2021-06-02:
You could use the Sensorbox 1.5 code as a starting point, but i guess you have that atleast partly working, otherwise the SmartEVSE will show an error message on the LCD.

qasimali-byte, 2021-06-03:
> You could use the Sensorbox 1.5 code as a starting point, but i guess you have that atleast partly working, otherwise the SmartEVSE will show an error message on the LCD.

Sorry @mstegen  i didn't actually understand. can you please explain more ?

qasimali-byte, 2021-06-03:
> Sensorbox 1.5 code

yes i am using the Sensorbox 1.5  version .but EVSE not showing the Current. How can i emulate the Current from the Software?

mstegen, 2021-06-03:
```
for (x=0; x<3 ;x++) {
                    Irms[x] = Irms[x]* CAL;
                    pBytes = (char*)&Irms[x];                                   // get raw 4 byte Double 
                    Tbuffer[n++] = pBytes[3];                                   // Send MSB first
                    Tbuffer[n++] = pBytes[2];
                    Tbuffer[n++] = pBytes[1];
                    Tbuffer[n++] = pBytes[0];                                   // Send LSB last
                }
```
This is the Sensorbox code that sends the measured current to the SmartEVSE. The measured current is stored in Irms[x].
Just change the Irms[0] to let's say 30.0 and you should see the data on the SmartEVSE.

qasimali-byte, 2021-06-09:
> ```
> for (x=0; x<3 ;x++) {
>                     Irms[x] = Irms[x]* CAL;
>                     pBytes = (char*)&Irms[x];                                   // get raw 4 byte Double [...]

hi @mstegen  i changed the code as you suggest me but i am getting same result . when i entered the STATE? commond still there is no state change information.
[image]
[image]
[image]

mstegen, 2021-06-09:
You asked "How can i emulate the Current from the Software" 
This seems to be working.. I see 2A displayed on the LCD for each Phase. Also the CAL values in the CLI show the measured currents.
If you want to see the set charge current with the STATE? command, then you first need to start charging. Connect a EV, or emulate one with a EV simulator (a couple of resistors and and Diode)

qasimali-byte, 2021-06-09:
okay. Thanks but my main concern with the STATE? command output. Could you please help me to simulate the Circuit so i can test the STATE command? like how many resistors required or Diode and with how much value etc?

qasimali-byte, 2021-06-11:
> You asked "How can i emulate the Current from the Software"
> This seems to be working.. I see 2A displayed on the LCD for each Phase. Also the CAL values in the CLI show the measured currents.
> If you [...]

hi, @mstegen I am using the Following circuit to emulate the EV and test the State? is that correct or need to rectify?
[https://www.evalbo.de/2015/02/14/typ-2-stecker-beschaltung/]
[image]