SmartEVSE/SmartEVSE-2 #13: ERROR NO SERIAL COM CHECK WIRING Should I use 120Ω termination resistors ?

henriterhofte, 2021-05-17:
I have the SmartEVSE running fw 2.20 connected to Sensorbox 2. New this weekend: (finally) also connected to the Eastron SDM630-Modbus V2 meter.

Functionally, everything works (charching, kWh counting, smart reaction to other loads to prevent MAINS overload)
but  I also get this error regularly:  ERROR NO SERIAL COM CHECK WIRING.

I used very short (10-15cm) non-twisted older phone cables with 4 non-stranded wire cores, hooked op 12V, GND, A and B for the Sensorbox-2 and hooked up A B and GND for the Eastrom 

Given these errors (which do not seem to hinder the functionality):

1) Should I use twisted pair wiring even for these short RS-485 connectrions?

2) I not only connected not only A and B between SmartEVSE and  Eastron, but also GND; is this correct?

3) Should I use 120 Ω termination resistors that seems to be recommended for RS-485 bus topology wiring?

PA1RB, 2021-05-17:
Modbus over RS-485 is specified for twisted pair cabling with characteristic impedance of 120 Ohm.

In practice it is often installed with Ethernet-grade  unshielded or shielded twisted pair, which has a differential impedance of 100 Ohm.

When resistors are mounted at both far ends of the bus (between the A and B wires), each matching the differential impedance, the reflections of signals are minimized.

For such short cable runs, and the system speed of 9600 bps however, reflections are unlikely to be a problem in your case. Also cable resistance is not likely a problem.

The SmartEVSE controller has no built-in resistors. Not sure about sensorbox-2.

Perhaps there is a strong source of interference nearby (e.g. switching power supply)? In that case twisted pair cable may help, make sure to use a single pair of twisted wires to carry A and B signals. Shielded cable, with the shield connected to protective ground may also help.

Perhaps a more likely source is a bad contact or occasional short circuit (with a single small copper strand)?

And yes, signal ground should be connected in parallel with the A and B lines in Modbus installations, although this is often omitted and the installation still works (be it by chance, not by design).

martijndierckx, 2022-01-24:
I have the same issue here.
My modbus setup/devices:
- Eastron SDM630
- SmartEVSE (2.30)
- Protoss PE11 (TCP interface)

If I just interconnect the devices, everything works.
I can even set the SmartEVSE to SMART mode and read the values from the Eastron.

But, once I start polling the Eastron for data over the PE11, I get the same 'NO SERIAL COM' error.
Code I use to poll the Eastron: https://github.com/martijndierckx/sdm630-modbus-tcp

Any ideas?