SmartEVSE/SmartEVSE-2 #15: Read / Write Modbus

TonKro, 2021-05-30:
I am trying ro read / write registers via modbus but the EVSE does not answer.

- Configured 8600 8 N 1
- Checked the bus if the data is there, which is
- Eg ty to read out register 1

Send out: [1,4,0,1,0,1,96,10]

Also tried as other Unit-ID like 2: [2,4,0,1,0,1,96,57]

No answer from the Smart EVSE.

When enable Smart / Loadbalancing. The data is there and working correct between the EVSE and Sensorbox.

What am I doing wrong?

bobosch, 2021-05-30:
Which version of SmartEVSE do you use?

TonKro, 2021-05-31:
Good question, the specific version I do not know, I did purchase it via stegen.com on the website like 1-2 Months ago assuming I got the most recent version by then.

mstegen, 2021-05-31:
When you enter the setup menu, the software version should be visible on the LCD.
The version is also shown when powering up the controller for two seconds.

TonKro, 2021-05-31:
Thanks.
Just checked, it is indeed v 2.20

bobosch, 2021-05-31:
Please use this modbus manual
https://github.com/SmartEVSE/SmartEVSE-2/tree/b8b6a1a2def22a2bfc3cc30be2f2ef639e2d4389#register-for-external-devices

Try to read register 0x00A0

TonKro, 2021-05-31:
Which Unit ID should be used?

bobosch, 2021-05-31:
It should be 0

TonKro, 2021-06-01:
As far as I am aware of could the unit id only from 1-255?

TonKro, 2021-06-01:
Maybe you can share a serial read and write command of a certain register of the EVSE?

bobosch, 2021-06-01:
Yes, 0 is broadcast address, in version 2.20 the load balancing option defines the modbus address (0=disabled, 1=master, 2=node 1....). will be fixed in next release.

TonKro, 2021-06-03:
What does the unit expect with read and write command on the serial bus?