SmartEVSE/SmartEVSE-2 #18: Smart mode

gransat, 2021-08-20:
Hi,
I have problem with smart and solar mode, I cannot get current below zero from current sensor. Current transformator is OK,
it gives +mV and -mV when it s clip oposide, but it is still +A, I can t get -A on controller.
Can you help me solve this problem?

bobosch, 2021-08-20:
Some electric meter use unsigned current values.
What device do you use for current measurement?

gransat, 2021-08-20:
I have original current transformers and sensorbox v 1.7 from stegen dealer. Controller is ver.2.20.

mstegen, 2021-08-23:
Only Sensorbox 2 (black enclosure) can measure the direction of the current. If you recently bought the Sensorbox 1.5 you can send it back and upgrade to the Sensorbox 2.

gransat, 2021-08-24:
Thanks for replay,

 

can you sen dme information, when will be controller available, on eshop is out of stock,

 

s pozdravom  Granec

 

GRANSAT s.r.o.

Košická Belá 595, 04465 Košická Belá

Tel.: +421905901086

IČO: 45371962

IČ DPH:SK 2022969784

 ***@***.***> ***@***.***

 <http://www.gransat.sk/> www.gransat.sk

closed 2021-08-29 by mstegen