SmartEVSE/SmartEVSE-2 #22: Advice how to implement charging over night

jimon, 2021-09-24:
Hi, I would like some advice how to implement charging during specific periods and/or remotely controlled.

A super naive way I can do right now is to MITM Sensorbox report and replace it with predefined frame that tells the controller "no energy is available", but that will result in my car complaining that charging cannot be started.

I could also expand solar timer logic to achieve something similar.

Which way should I take? Thanks!

jimon, 2021-09-24:
Guess the idea of sensorbox becoming the master device is better, than it could run this logic and control charging stations modes.

PA1RB, 2021-12-17:
@mstegen on the Sensorbox: _I'll release a version of the PIC firmware soon that will only do the CT measurements, and will not use the RS485 bus, nor use the P1 port. On this version the ESP32 will handle everything else._

Is this already on the horizon, or any support the team can give?

mstegen, 2022-02-10:
@PA1RB I've uploaded the code for the Sensorbox-2 to Github. It's still acting as a slave/node to the SmartEVSE, but all the modbus handling is now done by the ESP. Also added a way for the SmartEVSE to retrieve the time/date so charging at a specific time/day could be added to the SmartESVE code.