SmartEVSE/SmartEVSE-2 #23: Add modbus input validation?

ict-one-nl, 2021-11-07:
I'm still running the 2.17 firmware, but it does crash sometimes when modbus is used (integrated it myself using my https://github.com/ict-one-nl/Modbus2Mqtt library). Looking at the code - maybe I'm looking in the wrong place - but I don't seem to see input validation (expecting an int, make sure it's a valid int) on incoming modbus traffic. 

I suspect this could be the issue of my stability issues, in the 2.20 firmware I'm not seeing it either (but I'm not a C++ crack so bear with me :) )

mstegen, 2021-11-07:
Can you give me an example in the code? All modbus registers are 16 bit wide, what the purpose of this data is, depends on the modbus device. It can be a signed or unsigned 16 bit integer, the MSB or LSB of a 32 bit integer, or half of floating point number.

IgorYbema, 2021-11-12:
Isn't this just because you are using modbus itself wrong as a protocol? Your mqtt gateway as a master talking to smartevse as a master isn't supported in modbus. Bad timing and both master are active at the same time.