SmartEVSE/SmartEVSE-2 #24: Nodejs integration

martijndierckx, 2022-01-09:
Hi all

Just wanted to let you guys know, I'm working on an integration for Nodejs.
The first alpha version allows you to poll for all the values. It doesn't yet allow you to set them. That's for next weekend :-)

Link: https://github.com/martijndierckx/node-smartevse

martijndierckx, 2022-01-09:
I do have some trouble interpreting some of the Modbus registers. A bit more documentation on what each line does would come in handy.

kennyschuoler, 2022-01-10:
Hey Martijn, 

What more info do you need ? I can try if i can help you : I just wrote a first version of a custom integration for home assistant ( https://github.com/kennyschuoler/homeassistant-smartevse )

fluppie, 2022-01-10:
@kennyschuoler what HW interface do you have towards the SmartEVSE? Modbus over TCP?

kennyschuoler, 2022-01-11:
@fluppie i'm using a Waveshare RS232485 ( https://www.amazon.de/-/en/gp/product/B07RQSPYFV/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&th=1 ) ( mainly because this is located in my electrical box , otherwise you can also use the smaller RS485 version )
I have configured it currently with Modbus RTU over TCP , but depending on the config in Waveshare it is also possible to use Modbus over TCP ( i was able to get the values )

fluppie, 2022-01-11:
Is this module acting as a second Modbus master? Or is it invisible to the SmartEVSE and eventual kWh meters? Myself I have 2x Eastron SDM230 for house load and EV charged kWh's. In the future one for Solar could join the party.
Would a ESP32 be able to be a RTU to TCP bridge or is your software specifically made for the Waveshare module?

https://github.com/eModbus/eModbus

kennyschuoler, 2022-01-11:
To sIt is indeed acting as a second Modbus master, therefor i have to make sure that i only request data every +30 secs ( I am using the latest master branch software  ).

To give you an idea : My setup contains a sensorbox v2, 2 x Smart EVSE ( master + node ) devices and 2 x Eastron SDM630 modbus v2 devices.

The sensorbox is connected via RS485.

I also have solar panels with an energy meter, but this is completely isolated. 

The 2 eastron devices i have connected to 1 Waveshare device, and the other Waveshare device contains the 2 Smart EVSE devices and the sensorbox. They share the exact same configuration, so it should be able to connect them all to the same device.( just make sure that the eastron devices use an unused modbus address )

Hope this helps !