SmartEVSE/SmartEVSE-2 #25: Car plugged in + Car battery state

kennyschuoler, 2022-01-11:
I made an integration for Home Assistant but to have a complete automation i'm missing some information :

- Is there a possibility to detect if a car is plugged in ( which can be published via modbus ) ?
- Is there a possibility to read the EV battery percentage ? ( If i'm not mistaken this information is shared via the digital protocol, but i couldn't find any starting point of how this protocol works )

Could you please advice me how this can be achieved ?

Thanks !

martijndierckx, 2022-01-11:
0x0000 state should give you this info. B = Connected, C = Charging

kennyschuoler, 2022-01-11:
Thanks ! Then only 1 question remains :

Is there a possibility to read the EV battery percentage ? ( If i'm not mistaken this information is shared via the digital protocol, but i couldn't find any starting point of how this protocol works )

martijndierckx, 2022-01-11:
Afaik this is not possible. Certainly not in a standardised way. Some cars might publish this information on the CP/PP pins, but this would be in a proprietary way … I think.

But … maybe you could achieve the same via a cloud api of the car brand?

fluppie, 2022-01-11:
Nope, mode 3 chargers are not able to read the EV battery it's SoC. If I'm right, mode 4 chargers can, like these DC chargers.
Indeed, using for example BMW Connected drive etc.
[image]

kennyschuoler, 2022-01-11:
Aaah to bad, thanks for the information ! I did not know that only mode 4 chargers are able to share that information. That explains why i couldn't find any information :p

I will indeed then use the BMW API in this case, i hope the information is near real-time while charging ( i'm using the values to keep a charging history )

Thanks @martijndierckx and @fluppie !

kennyschuoler, 2022-01-11:
Maybe a small addition, there are however mode 2 chargers available, which can display the battery charging state : 

[image]

Any idea how that works  ?

kennyschuoler, 2022-01-11:
Maybe it is possible based on the Pilot voltage ? But then it seems it should be possible ( but it is really not accurate )

[image]

Reference : https://batteryuniversity.com/article/bu-903-how-to-measure-state-of-charge

mstegen, 2022-01-11:
The SOC of the battery is not readable from the EVSE. There are however way's the EVSE can detect that the battery is almost full. Most cars will taper down the charge current when they approach 100% SOC. 
This is detectable with charging equipment. 

This only works when charging at high charge current, and when charging to a high SOC level. And the tapering will only happen in the last minutes, so in practice it's not very useful. 
Maybe the above EVSE has a built in energy meter to count the charged energy, and it can estimate the SOC this way?

fluppie, 2022-01-11:
https://en.wikipedia.org/wiki/Combined_Charging_System
Charging communication

Unlike the connector and inlet, which depend on the geographical location, the charging communication is the same around the globe. Generally two types of communication can be differentiated.

Basic signaling (BS) is done using a pulse-width modulation (PWM) signal which is transferred over the control pilot (CP) contact according to IEC 61851-1. This communication is used for safety-related functions, indicating for example if the connector is plugged in, before contacts are made live (or energized) and if both charging station and electric vehicle are ready for charging. AC charging is possible using the PWM signal only. In this case the charging station uses the duty cycle of the PWM to inform the onboard charger of the maximum available current at the charging station (A pulse width of 5% indicates that HLC shall be used).
High-level communication (HLC) is done by modulating a high-frequency signal over the CP contact (also known as Power Line Communication or PLC) to transfer more complex information, which may be used e.g. for DC charging or for other services such as "plug and charge" or load balancing. High-level communication is based on the standard DIN SPEC 70121 and the ISO/IEC 15118-series.

https://en.wikipedia.org/wiki/IEC_61851

Ik denk via de API nog het makkelijkste :-) ?

martijndierckx, 2022-01-11:
I think it’s a proprietary protocol which allows this. But if available, it will be different for every car brand.

kennyschuoler, 2022-01-11:
API it is indeed ! :p Thanks all !

closed 2022-01-11 by kennyschuoler

fluppie, 2022-01-16:
I found this regarding Charging Protocols: https://www.openchargealliance.org/downloads/