SmartEVSE/SmartEVSE-2 #26: No response for 'state' over Modbus on FW 2.30

martijndierckx, 2022-01-11:
Hi

Just updated my SmartEVSE to 2.30.
Where I was able to get the state from address 0x00A0 on FW 2.13, I'm not able to get in from 0x0000 on FW 2.30.
Other addresses seem to work fine.

Any ideas?

Kind regards
M

kennyschuoler, 2022-01-11:
I have the same issue it seems ( @martijndierckx thank you for your previous remark :) ). 
Tomorrow i'll compile the latest version and check if it is already fixed in the meantime ( The charging current is also not working as it should , but looking at the commits that should be fixed )

mstegen, 2022-01-11:
Please note that the modbus registers changed between v2.30 and <=2.20 

use this list for 2.20 or earlier:
https://github.com/SmartEVSE/SmartEVSE-2/tree/b8b6a1a2def22a2bfc3cc30be2f2ef639e2d4389#modbus
and the default one for v2.30:
https://github.com/SmartEVSE/SmartEVSE-2/tree/b8b6a1a2def22a2bfc3cc30be2f2ef639e2d4389#modbus

martijndierckx, 2022-01-11:
I’m using the new list. Everything seems to work except the first state address.

https://github.com/SmartEVSE/SmartEVSE-2/tree/b8b6a1a2def22a2bfc3cc30be2f2ef639e2d4389#modbus
https://github.com/SmartEVSE/SmartEVSE-2/tree/b8b6a1a2def22a2bfc3cc30be2f2ef639e2d4389#modbus

kennyschuoler, 2022-01-11:
@mstegen It is a possibility that i have a slightly different issue then @martijndierckx , if it is the case i'll create a new issue

I am also using the latest registers - see screenshot ( Everything else is working as expected - ignore the the callibration value, i still need to display that one correctly )
- The charging current is 160 A ( while it is not charging atm, this is already fixed based on the commits )
- The state that is returned are numeric values ( i suppose 0=A, 1=B, 2=C, ... )
- The state is always 0 except when i set the access bit to 1 ( allowed ) and i switch at 22h from solar to smart, then the state is set to 2 ( but it seems i don't get the status 1 ). When the battery is full, i set the access bit back to 0 . Then it seems the state is shortly set to 1
- At 7h in the morning, i set the access bit again to 1 and switch from smart to solar, then it also shortly shows the state=1  ( plugged in ) but then it returns to 0 
  --> In case this is not working as expected, it'll probably have to do something with the access bit and the state update
 
[image]

[image]

rvdgaag, 2022-02-17:
@kennyschuoler  How did you manage to read the values of the Smart EVSE ?

I have also fw 2.30 installed but can't get any values via modes from the smartens, both SDM630 eastern units reply fine.

Any suggestions ??