SmartEVSE/SmartEVSE-2 #32: How to enter to bootloader now?

gxnse, 2022-01-23:
Hello. I deleted this in the EVSE.c file:

```
            if ((timeout == 0) && !(Error & CT_NOCOMM))                         // timeout if CT current measurement takes > 10 secs
            {
                Error |= CT_NOCOMM;
                setState(STATE_A);                                              // switch back to state A
#ifdef LOG_WARN_EVSE
                printf("\nError, communication error!");
#endif
                ResetBalancedStates();
            } else if (timeout) timeout--;
```

and I think this is the reason why I cannot enter bootloader mode. It says "bootloader not found". Screen always stays bright and each minute it gives error about temperature. So as far as I understand it thinks that there is a car connected and does not give opportunity to enter bootloader mode. What should I do now? I want to reprogram it back as it was... :(

gxnse, 2022-01-23:
I just can't enter bootloader now... :(

gxnse, 2022-01-23:
[image]

mstegen, 2022-01-24:
What is causing these temperature messages? Is it really that hot inside?

You can try this:
Make sure nothing is connected to 12 pin connector.
connect the FTDI cable to the SmartEVSE.
start the AN1310 (bootloader program) and press the break(pause) button (F3)
power up the SmartEVSE.

It should now enter the bootloader, if not you might need to reprogram it using a pickit.

gxnse, 2022-01-24:
It is not hot inside.

It does not enter bootloader. I guess I need pickit now. Is microchip is soldered there or I can just take it off?

mstegen, 2022-01-24:
Yes you can try it with a pickit, then you can program the complete pic, including the bootloader.
You can also send it back to me if you want, then i'll reprogram it for you.

gxnse, 2022-01-24:
You can email me on mainemail.exe@gmail.com

mstegen, 2022-01-31:
I'm also located in the netherlands, just send it to the address listed on www.stegen.com

closed 2022-01-31 by mstegen