SmartEVSE/SmartEVSE-2 #33: HW fix for LCD re-init?

zharijs, 2022-01-28:
Hi.
What's with the re-init of LCD at contactor open and close functions? Surely, this is a quick patch not a solution. I'm volunteering to fix the HW issue causing this to be needed, if someone can tell me how exactly to reproduce the LCD issue on my side. What contactor was used, what ware the load conditions, how long ware the wires?

[image]

mstegen, 2022-01-31:
Hi, 
The re-init was added to fix garbled LCD screens. It sometimes happens with big contactors that are next to the SmartEVSE.
Even when there is no load connected to the contactor it sometimes garbles the LCD.
I've also seen it happen when the contactor switches a big 2 phase to one phase transformer on or off.

zharijs, 2022-04-19:
Hi.
Yup, that's what I thought, but I could not reproduce it. 
Perhaps you have P/N for the contactor that caused it? I could order it then, add hardware fix and confirm that it would not happen anymore.

AndrejValand, 2022-05-25:
I also had some problems with display sometimes not working after contactor switch off/on. I have done extensive testing with an old nasty contactor. Below the HW+SW updates to make the whole LCD functionality much more robust. With this updates i could not bring the smart EVSE to have problems with the LCD anymore.

Hardware Modifications:
[image]
[image]

Needed Firmware modifications GLCD.c:
[image]

Needed Firmware modifications EVSE.c:
[image]
[image]
[image]

mstegen, 2022-05-27:
Hello Andrej,

Thanks for your suggestions, i'll most likely implement your changes in the next HW rev of the LCD board.

AndrejValand, 2022-05-29:
Sure, glad to be of help to the project. I have tested and you can also simulate the LCD corruption issue with the standard iskra contactor but using a much longer FPC cable and a moving the LCD board out of the SmartEVSE housing. I know its not a real situation, but it simulates the increased influence of external EMF on the LCD system of the SmartEVSE than can occur in some situations.