SmartEVSE/SmartEVSE-2 #34: Switch between single phase and 3 phase

martijndierckx, 2022-01-28:
In theory ... would it be possible to switch between single phase charging and 3 phase charging using 2 relays? Would it possible to control those 2 relays, instead of the default 1 using the current SmartEVSE2 hardware, by changing the firmware?

Let me elaborate:

I want to make my SmartEVSE even more smart by reading the battery status from the car (not via SmartEVSE, but directly using the cars' API). And based on that, calculate what the most ideal charging current would be.
Why? Because in Belgium you get charged based on the current peaks you cause on the network, so you benefit by keeping the charging current as low as possible.

The minimum charging current is always 6A:
- On a single phase: 6A x 230V x 1 = 1380W
- On three phases: 6A x 230V x 3 = 4140W
This could have a big impact on the monthly energy bill.

So if I calculate that 6A or 10A single phase charging would be enough to charge the car overnight, I would tell SmartEVSE to only use a single phase (over Modbus), which would then only switch on relay 1 controlling phase 1.
If I calculate that multiple phases are needed, I would again instruct SmartEVSE to do so, which would then switch on relay 1 for phase 1, and relay 2 for phase 2 and 3.

When switching between single and three phase charging, SmartEVSE would need to pause charging for a couple of seconds, because some vehicles might not handle the switch between single and multi phase charging well.

Your thoughts please...

fluppie, 2022-01-29:
Pretty good idea actually. A second contactor could provide this functionality. Or use 2 single phase contactors for 3 phase instead of 3phase contactors. Maybe a seperate microcontroller with modbus communication that controls this contactor? Then the SmartEVSE only needs a software update and not a HW revision.
Something like this https://webshop.domoticx.nl/index.php?route=product/product&product_id=1868 but with decent relays (I just would use another contactor)

mstegen, 2022-01-31:
Yes, two contactors to switch between 1/3 phase charging is a good idea. 
My main idea was it to use it when solar charging. You start slow with only one phase, and once you get to 16A(or 18 A) on one phase, you switch to tree phases @ 6A.  But it could also be used for the situation in Belgium ofcourse (never thought about that!) 
I think it's a bit difficult to add it to the SmartEVSE v2 code, as there are no unused IO pins anymore that can be used for this.

And you also need a way to disconnect the Control Pilot signal from the EV, so it thinks it's really unplugged. 

Here's a picture of the SmartEVSE v3 prototype: 
[image]

martijndierckx, 2022-01-31:
Nice!
ETA on v3?

mstegen, 2022-01-31:
They are being manufactured now. ETA is February.

WilfredHoogerbrugge, 2022-01-31:
I've been thinking about this too, nothing new actually the 'Zappi' already does this (and had some serious issues with the e-Niro and Kona, see: https://myenergi.com/zappi-2-firmware-version-3-033/ , scroll down for a full description of the problem)

Would it be an option to use the ouput for the motorized socket lock for the 2nd relais so owners of the current (v2) version can also use this featue? Ofcouse this only works when no lock needs to be controlled, like in my situations where a fixed cable is used.

mstegen, 2022-01-31:
Using the lock output to drive a (solid state) relay, which in turn then can switch the relay on/off could work. But you still need a way to isolate the CP signal. A second relay between 12V and the second output might be used for this. As long as the coils don't draw to much current it could work.

prjh, 2022-02-02:
That are interesting good news. An by the way I saw an USB-C socket on your prototype picture. 
... conform to upcoming EU regulations :-)
THX