SmartEVSE/SmartEVSE-2 #35: Smart charging not working as expected due to latency

poelie83, 2022-02-05:
Hi all,

After charging for some while I started checking the system in Smart mode.
I have the SmarEVSE-2 and sensorbox 2 connected with 20m Cat6 cable, the sensorbox uses the P1 port to receive data.
The Green status led is blinking every 2 seconds when data has been received.
All above looks fine, until the load balancing needs to kick in. 

The update rate from the sensorbox to the SmartEvSe-2 is way to slow. measured latency from disconnecting or connecting load is almost 10 seconds!! Due to this latency the SmartEVSE-2 is not throttling the output correct en go's to wait for power mode.

To make sure the update rate is fine I checked the RAW data with domoticz and a USB P1 logger, the update rate is 1 second.
Any ideas to fix latency issue?

mstegen, 2022-02-06:
Hi, 
This should not happen, Every 2 seconds the SmartEVSE will receive a measurement from the Sensorbox, and if one of the phases is measuring above the Mains current (usually 25A in NL) it will immediately lower the charge current of the EV. 
The measured phase currents are always visible on the LCD of the SmartEVSE.

You can check if everything is working correctly by switching on/off a load, and observing the values on the display.
They should change with a delay of max 2 secs.

The reason for different behavior could be a non compliant Smart Meter. I observed this when they installed a Landis+Gyr ZMF110CCtFs2 
This smart meter does send a P1 message every second, but the data (except for the timestamp) will not change for 10 seconds!!

poelie83, 2022-02-06:
After your message I started to do some more research, I also live in the Netherlands, main fuse 3x 25A, the smart meter is a Landis+Gyr E350 with DSMR 5.0. update rate 1 second with exact the same behavior as described above.
The smart meters sends a P1 message every second but the data is only updated every 10 seconds :-( 

Is it possible to add some Landis+Gyr support in you product? instead of scaling down every 2 sec to a lower power mode I suggest to go to the lowest power mode and step up every 10 sec until the 25A has been reached? this is the safest way to prevent fuse blow out. it would be sad to add three current measure clamps to resolve the meter issue.

Thanks for the quick reply!

mstegen, 2022-02-06:
Ah, yes the E350 is unfortunately not supported. Weird that it works, i have never seen a E350 with a (D)SMR 5 telegram.

For what it's worth, i told the 'netbeheerder' that the installed meter is not DSMR 5.x compliant, and they exchanged it for a Iskra AM550-TD2, at no extra cost.

The SmartEVSE code requires a accurate measurement every two seconds, it's not easy to add code changes that change this to 10 seconds, and imho 10 seconds is too long if there is a sudden power increase.

poelie83, 2022-02-07:
I will try to contact the netbeheerder first :-)

please see the photo attached with the SMR 5 value on it, thanks for the quick reply! love your work.

[image]

mstegen, 2022-02-07:
It's the same as the one i once had. The ZMF110CCtFs2 model, look at the top left corner..

poelie83, 2022-02-10:
22th of February the "Netbeheerder" will swap the meter :-) I will post an update after the swap.

IgorYbema, 2022-02-10:
Anyway, the fuse will not blow up that fast. It can take that more load for a few seconds or even minutes. So this 10 second delay, although weird, is no problem. DSMR4 meters update each 10 seconds also and you wont see blown fuses there.

mstegen, 2022-02-10:
@poelie83 Great news! 
@IgorYbema It would require some code changes to the SmartEVSE software, in order to make this work on this smart meter. The SmartEVSE will increase/decrease the chargecurrent every 2 seconds. If it measures for example 27A at one phase, and the max MAINS setting is 25A, it will lower the chargecurrent to the EV with 2A. 
With this L+G meter it would lower the chargecurrent over a period of 10 seconds, 5 times with 2A. so in total 10A.
Obviously this will not work correctly.

IgorYbema, 2022-02-11:
Ok I get it now. And forgot smartevse doesn't support DSMR4 :)
Although 10 secs would still be ok for most fuses I rather would keep it on 2 secs too :)