SmartEVSE/SmartEVSE-2 #36: SmartEVSE not recognising car anymore

RIOuShil, 2022-02-07:
Probably hardware issue, not software issue. But want to rule out software before swapping hardware out.
SmartEVSE was always working without issues. Great product. I have 2 in load balancing mode. And an ESP8266 connected to the smart meter and my kwh meters. ESP8266 runs modified sensorbox code to provide the EVSEs with the current numbers and update my smarthome system with readings.

Recently 1 of them started malfunctioning. I plug the EV in, but EVSE does nothing. Sometimes the LED screen will light up, sometimes not.
Most of the time it worked again after turning the EVSE on and off. But even that is now no longer working.
If I put the multimeter on CP and GND, it sometimes shows around 9V, sometimes more around 6V. If I then turn EVSE on and off, it works mostly. Putting the multimeter on there, somehow makes it sometimes work after turning EVSE on and off. 
If EVSE turns to smart charging, it typically stays active for quite a while. It might drop every once in a while for unexplained reasons (ie enough power available), but then usually picks it back up again soon

So my guess is a hardwire issue on GND or on CP. Before these problems started, I had quite some unexpected serial comms issues, which went away when I simply disconnected GND. I have the 2 EVSE in load balance. 

Any thoughts on how to fix welcome. I might swap the 2 EVSEs to confirm it is really the EVSE (and not the fixed cable connected to it), but have not had time to do this yet and would rather not if I did not have to

mstegen, 2022-02-07:
Hi, 
Could you please let me know the software version the SmartEVSE's are running? 

When there is noting connected to the CP pin, it should measure +12V. When a car is connected, it's no longer possible to measure it with a multimeter as it will vary between +9V/+6V and -12V at 1000 times a second.

RIOuShil, 2022-02-07:
Hi! 
I'm on 2.13.1. Was always working well, so did not need to update (and if I update now I would need to adjust my ESP8266 code, because the Modbus addresses changed I believe)

Indeed, 12V, 0Hz, when nothing correct. I get 9V 1000Hz when cable is connected, but nothing happening. 6V 1000Hz when charging (I think, cannot get it to charge now).

Tx for your help!

p.s. If I need to get a new unit, then totally fine. It has worked reliably for long time and it is a great product that I highly recommend :)

fluppie, 2022-02-07:
@RIOuShil would you care to share your ESP8266 code, since a lot of people try to get this functionality :-).
I assume all modbus meters are connected to the SmartEVSE and you kinda read them out in parallel or by sniffing the RTU modbus line?

RIOuShil, 2022-02-08:
Sure @fluppie. I'll see if I have time over the weekend. But honestly, I'm a really bad coder. Fully self taught. So my code is very very crappy. Specifically the bit bashing of Modbus was quite a challenge for me. But, it works for me now, so happy to share (but with a big disclaimer). I do think it would be wise for @mstegen to release his upgraded sensorbox, with for sure much better code :)

mstegen, 2022-02-08:
When at +9V/-12V 1kHz, the EV should detect the signal, and initiate the charge by pulling it to 6V. 
Have you checked the charge cable/plug for water ingress?

Ofcourse i can always check the smartevse for you on my test bench. Just send it back to me. I'll check/repair it if needed.

RIOuShil, 2022-02-08:
tx. Let me do some more tests first indeed. Easy would be to swap the 2 EVSEs. That would rule out the cable as issue. If cable is the issue, it is probably the plug I guess? Do not really think water could get into the cable itself. I tried opening the plug before, but could not fully reach all the torque screws and it seemed rather well build, so was kind of assuming water would not get in. But I can buy some longer torque screwdrivers and have another go for sure. tx for the help!

IgorYbema, 2022-02-10:
> @RIOuShil would you care to share your ESP8266 code, since a lot of people try to get this functionality :-). I assume all modbus meters are connected to the SmartEVSE and you kinda read them out in parallel or by sniffing the RTU modbus line?

@mstegen released the esp32 sensorbox code today. So you can add to that for your own monitoring purpose

RIOuShil, 2022-02-15:
Quick update: It is probably water ingress in the charger. I have one with a button on top to open the chargeport. Button should be waterproof. However, the PP to Earth resistance should be 220 ohms, but when measured is way too high and all over the place. Contacted the vendor and they are sending a replacement. Useful guide here (outlines how to measure resistance, etc.):
https://www.evchargeking.com/Media/Uploaded/Tesla%20Connector%20Installation%20Instructions.pdf