SmartEVSE/SmartEVSE-2 #39: Solar charging stability

mintgroen, 2022-02-27:
I noticed possible undesirable behaviour of the available current calculation.

[image]

My Hyundai Kona connected and the SmartEVSE set to smart charging, with max 6A import set and timer at 60min. Solar panels on phase 1 and 3 are happily producing and solar charging started a 4A export.

At 10:47 you see the variation in power drawn on all phases starting. Looking at the SmartEVSE display, I saw the available current flipping from 7A, to 6A (timer started) and back to 7A again and flipping to 6A again.

[image]
[image]

At 10:52, I've manually changed the import to 5A to stop the switching.

Is this behaviour intended and ok? Or should there be a stabilising functionality that avoids the switching back and fort every 10 seconds?

mintgroen, 2022-02-27:
After some minutes, the value started flipping again. I've set the max import back to 6A at 11:09, variation increases.

[image]

Why does this happen?

mstegen, 2022-02-27:
It could be that the car is cold, and that it tries to pre-heat the battery. 
On my model S i see something similar. When cold, it keeps switching between 6-8A, and after 30 minutes or so, it stabilizes and starts charging normally.
The SmartEVSE will vary the charging current in 0.1A steps, maybe the Kona accepts only steps of 1A (just a wild guess here)

mintgroen, 2022-02-27:
It's the EVSE that is throttling back and forth:

https://photos.app.goo.gl/iqvHoNuMvBxxS5LX7

But why?

Setting the import to 0A make the current stable again.

bobosch, 2022-02-28:
Do you use modbus to get the measurements for the nice graph? Maybe to many collisions

mintgroen, 2022-02-28:
The measurements are p1 telegrams, the same as the sensorbox receives. (Which is blinking green)

mstegen, 2022-03-01:
I checked your video, the EV is not responding quick enough to the current set by the EVSE (or the SmartEVSE is increasing the current too fast)
Software version 2.3 increases the current with max 0.5A per 2 seconds, when solar charging. This should help stabilize the set current.
Ideally it should depend on if you are charging on 1 or 3 phases, but this is not implemented yet.

https://github.com/SmartEVSE/SmartEVSE-2/releases/tag/version_2.30
If you do not have a programming cable to update the controller to version 2.30, you can send it to me for reprogramming.