SmartEVSE/SmartEVSE-2 #40: Unable to read modbus on V3

k-janssens, 2022-03-10:
I've installed the latest V3 yesterday and the hardware is working great.
Only problem I currently have is to read the modbus registers.

Regardless if what register I try to read, I always get the exact same response:
`21:04:19:482 > 00 49 00 00 00 06 01 03 00 02 00 01 
21:04:20:509 < 00 49 00 00 00 06 0A 04 00 00 00 14 
21:10:18:578 > 00 4A 00 00 00 06 01 03 00 04 00 01 
21:10:20:513 < 00 4A 00 00 00 06 0A 04 00 00 00 14 
21:10:47:492 > 00 4B 00 00 00 06 01 03 00 0A 00 01 
21:10:48:516 < 00 4B 00 00 00 06 0A 04 00 00 00 14 
21:10:52:844 > 00 4C 00 00 00 06 00 03 00 02 00 01 
21:10:54:510 < 00 4C 00 00 00 06 0A 04 00 00 00 14 `

Besides the response always being exactly the same, the value '0A 04 00 00 00 14' can be found on the Sensorbox documentation (see the example below the page):
https://github.com/SmartEVSE/Sensorbox-2 

So am I doing something wrong? Or is the modbus implementation changed in V3?

Additional info:
- tried both modbus and modbus rtu
- sensorbox is connected with the P1 of my digital meter
- no kWh meter
- I use a waveshare device to read modbus over tcp

mstegen, 2022-03-10:
Modbus implementation has changed. For v3 i'm using the eModbus library (https://github.com/eModbus/eModbus) which works different from the implementation from v2. 
But it works fine as master or as node controlled by a master, also mixed with older v2 controllers, but i have not tried to control it from a external modbus controller.

I'm trying to understand what response you are receiving, but i can not make any sense of it yet.. All i know is that a modbus message should not start with '00' 
The SmartEVSE uses modbus RTU @9600 bps 8N1
Maybe you should try to listen to the modbus data first, you should be able to receive measurement data from the sensorbox every two seconds.

k-janssens, 2022-03-10:
I'm using the same hardware as a friend of mine uses with the V2.
Got all his settings and such, but if he makes a request he is getting the proper response but I always get the same response: 00 00 00 00 00 06 0A 04 00 00 00 14 

No matter which register I try to call or any other setting. 
A few screenshots from the client I'm using:

[image]

[image]

In the Waveshare interface this is the data he is receiving every second (which make sense as the sensorbox is working as it should):
[image]

kennyschuoler, 2022-03-11:
I just replaced my v2 controllers with the new v3 controllers, and i have the same issue, i'm unable to connect to modbus anymore.
I also tried if we could directly connect to modbus via the wifi interface, but it seems this is not (yet) available ? ( It's also possible that it is not the default 502 port, but here i'm waiting for the source code to figure this out )

9600, 2022-03-11:
@mstegen You called?

kennyschuoler, 2022-03-11:
> @mstegen You called?

It was on accident ... He mentionned the bitrate to connect to modbus, you can ignore this ;p

mstegen, 2022-03-11:
Two modbus masters do not (currently) work on v3, this is mainly because the eModbus library does only support two modes, master or node/slave. When configured for master, there is no modbus address set
There is also an option for a TCP bridge, but i have not used that yet.
My main focus was to get communication from the master controller to nodes/ kwh meters and sensorbox to work.