SmartEVSE/SmartEVSE-2 #42: "Bootloader not found" revisited

cvanlith, 2022-03-30:
Just received a new sensorbox2 but couldn't get is to communicate. Therefore I tried to upgrade the software from 2.04 to 2.30. Unfortunately I get "Bootloader not found" when I try to connect  to the SmartEVSE module using a FTDI cable and an1310ui.exe. Normal CLI works.

I tried the suggestion from @mstegen:
* Make sure nothing is connected to 12 pin connector.
* connect the FTDI cable to the SmartEVSE.
* start the AN1310 (bootloader program) and press the break(pause) button (F3)
* power up the SmartEVSE.

But this also fails.
I never upgraded the unit before. What to do?

mstegen, 2022-03-30:
Is the FTDI cable connected the right way? The black wire should be on the side where it says 'BLK' on the circuit board.
Also make sure the correct COM port is selected in the setup of the AN1310 bootloader program.

If it still does not work, please send it back and i will reprogram it for you.

-Michael

cvanlith, 2022-03-31:
OK, I will order a new one and send the old one back after I have installed the new one. I need to charge my car :-)

closed 2022-03-31 by cvanlith