SmartEVSE/SmartEVSE-2 #43: Implementing a clock and schedule for minimum morning charge when in Solar mode.

scspijker, 2022-05-02:
### Description
Context: My car has a very large range (400km) and I only need about 80% (60kWh) per week to remain full.

I like the solar mode on SmartEVSE, only charging solar, I've though of a further improvement though.

Currently I just leave it in Solar mode all the time, and plug in everytime I park the car. This usually fills up the car (limited to 80%) to more than I need. Sometimes however, we have little sun for a lot of days, and my charge creeps down to 15%. Then I need to manually tell the charger to charge from grid, however, usually I think of this too late. I'd like to wake up in the morning to a minimum charge of 40%.

I think implementing a clock, and a rudimentary schedule would solve this issue. For example:

Set to solar mode
Set minimum morning charge level to xx%
Set minimum morning charge start time: 02:00 (after sundown, guaranteed in my place of residence).

Say the sun doesn't provide enough power to get my car up and over 40%, the SmartEVSE will start charging at 02:00 from grid power up to xx%, and then stop (unless solar power is available again).

This will make solar mode much more attractive during periods of low sun.

### TL;dr

SmartEVSE in solar only mode.
Sun does not shine enough, car not charged above 40%.
SmartEVSE will start charging using grid power at night until 40% has been reached.

scspijker, 2022-05-02:
I'm an programmer myself (originally embedded), and am looking to buy a new charger. Do you think it would be possible for the SmartEVSE to implement a clock, and this schedule charging?

If so, I might buy a V3 and open up some pull requests 😁

hyperbart, 2022-05-04:
This can be handled by using EVCC:
https://github.com/evcc-io/evcc
--> https://github.com/SmartEVSE/SmartEVSE-2/issues/44

scspijker, 2024-02-28:
This may be possible with new OCCP 2.0 standards

zharijs, 2024-03-24:
Throwing out some ideas -

1. I'm looking into implement OCCP in ESP - it's not trivial but seems possible. It would not help much with getting car data though, unless you have some connection from car to manufacturer and manufacturer would share the data through OCCP on internet. 

OCCP is is more for controlling, configuring and "multiboxing" the charger network.

2. Also I'm looking in to possibility of implementing PLC (as in power line communications not programmable logic controller) to read the charge state and vehicle statuses form vehicle directly. 

You need a special hardware for that. Interface chip / module manufacturers are quite secretive about it but there are some modules available it seems. One I red a bit about was bridging PLC to UART. So ESP debug serial multiplexed by minimal glue logic, switched by USB voltage (I've done that with single SOT23-6 package for ESP8266 before) could be used to talk to it. 

If anyone is interested in participation getting this implemented, I can share more of my findings in the field. 

3. About the schedule (RTC) - it should be simple fairly simple to implement system clock based RTC with periodic interrupts that would be set and corrected ever so often by NTP. Again - done that on 8266 before. The painful part would be to make the actual interface to configure the thing. 

4. Would it not be better to get the "charge authority" by reading/scraping Nordpool instead of preset RTC if you have your rate tied to exchange?