SmartEVSE/SmartEVSE-2 #45: Renault ZOE ZE50 start of charge problems on hot days

AndrejValand, 2022-07-27:
Hi,

I'm having problems with my ZOE in this hot days. When it enters the state C with the preset max current the car interrupts the charging to start up its heat pump assisted battery/charger air cooling system. SmartEVSE goes into a wait mode the car start up the cooling but depending on temperature it shuts it cooling off before the charging is started so this sometimes goes on and on... Charge start/stop/restart...

I found that if charging starts with the minimum preset current and waits a few seconds so the car can start all its internal cooling system than i ramp up the charging current all works well. It goes normal to full charging power without any issues. Slow ramp up of charging current is also beneficial for regulation and fuse safety in a home. In an overcurrent situation the current is reduced instantly but than it ramps up slowly again.

I made this changes to the code and i'm testing this now:
[image]

[image]

There is maybe some other more correct/elegant way of doing this, and i have no way to test this in a solar situation, but this slow current ramp up can be beneficial also for solar setups.

I also noticed that some better public chargers perform a slow charge current ramp up.

Any ideas and comments welcome.

Regards, Andrej

bobosch, 2022-07-28:
Please check out the latest git version. Two features where added recently that may solve this issue, too:
- At solar charging charge at least 40 seconds at minimum current (https://github.com/SmartEVSE/SmartEVSE-2/commit/0890c6872419c7850775f022a8d8cf7788768683)
- Avoid increasing current while charging at minimum current https://github.com/SmartEVSE/SmartEVSE-2/blob/901012e0d9319b2277556e6a8c6c9071016e9b1e/SmartEVSE2.X/EVSE.c#L1070

AndrejValand, 2022-07-28:
Ok will test this, but need to wait until its over 35°C again and have the car a few hours in the sun:) Tried today a few times as it was 27°C and the car did not even start the cooling until over 20min into the charging session and even then only the fan without the heat pump. In this normal situation it does not stop the charging to do this. Its interesting how this 8°C makes such a difference how the car works with its cooling system. But yes if the charging in this last version starts at minimum current and waits a few seconds, so the cooling system can engage,  than goes to max set current or whatever is available it should work. 

Will report back when i can test this last release on a hot day. Must i set the SmartEVSE to Solar mode? Will this test ok  without the solar installation? I use Smart mode now.

I also went today to a  few public charging stations in vicinity and all take quite some time before full charging is engaged 10-20s. Don't know it they are negotiating so long or are they starting with a low current then switching to full. Definitely one i tested i can hear it ramps up the CP current slowly (Renault Caméléon® charger you can hear as it used the motor for power conversion).

bobosch, 2022-07-29:
The "slow start" is only active on Solar mode, because there it's most likely that the car can only charge with a low current.

At Smart charging more current should be available. What is your Max current setting?

The ZOE uses 3 phases, up to 32 A, so 22 kW charging capacity. Maybe the heat pump cools the battery to allow faster charging only when a certain power level is available. Could you please describe the issue with some power levels or take a film from the SmartEVSE display? What do you mean by "wait mode"? On Solar charging with bad configuration (e. g. START 1 A and STOP 1 min) it will also switch on and off every minute when not enough current is available. On Smart charging it depends on your installation - how much capacity is available?

AndrejValand, 2022-07-29:
I have 20A main fuses in our house, and have 16A max current set for charging, 6A as the min current. Even if i set max current to 8A and if its really hot outside the ZOE ZE50 i have will not initiate charging with the SmartEVSE the first time. The only start current that it accepts in this situation not to disengage charging to start up its internal cooling procedure is 6A.

Initial condition is there is more than 16A available from the house, outside its 30°C+++ the car is in the sun. Smart charging is enabled. With 35°C outside this happens always (last week).
What happens:
-I connect the car, negotiating begins
-SmartEVSE sets the current to 16A the car wants to start the charge, but it disconnects in a second. I can hear the heatpump starting up. 
-SmartEVSE is waiting for a while, the car internal cooling is wining up to operation. In best situation the SmartEVSE restarts charging while the cooling is still active and the charging starts normally the second time. Sometimes the car internal cooling is not on long enough and switches of when SmartEVSE restart the charging procedure, so again the car disconnects the charging and starts up its internal cooling. This can go on for a while sometimes, depending on temperature, SmartEVSE contactor on/off cars internal heatpump on/off...

What i tested:
-I first set the maximum charging current to 6A on the SmartEVSE, and it worked without problems, engaging the charging the first time always even at this hot temperatures.
-Then i manually went into the SmartEVSE menu and increased the max charge current slowly to 16A and the car normally started the cooling and continued the charging procedure without interrupting it. 
-I did not test what happens if i start the charge at 6A and then after say 10s  change it immediately to 16A.
-I tried some public AC charging stations and they worked even at this high temperatures the first time
-Next i added the code i posted above and it was also working fine with SmartEVSE (starting at 6A, waiting a bit than ramping up to full current). But as current is changed and calculated at few places in code i'm not 100% certain the solution is 100% compatible for solar and multi SmartEVSE solutions.

I can make a video next time, for sure there will be this extreme hot days in august.

In the solar charge situation slowly ramping up the current could maybe also reduce on/off situations, but i don't have a solar installation to test this. Also i don't know if some other cars could have a problem if charging is started with 6A and then ramped up to full, but i think not.

bobosch, 2022-07-29:
Solar mode won't work without PV ...

I've created a patch to enable the change also in Smart mode and changed the Phase detection to start 10 seconds with 6 A charging. Please apply it and check with this version. After 40 seconds it will try to detect the number of phases used by changing the current between 6 A and 8 A, after this the current will increased.
[delay.diff.txt](https://github.com/SmartEVSE/SmartEVSE-2/files/9218210/delay.diff.txt)

AndrejValand, 2022-08-14:
Hi,

There still was no day that was over 35°C++ in the last two weeks so i made three videos in the mean time: 
 - SMART charge with phase detection charge engagement
 - SMART charge with current ramp up
 -  Standard SMART charge charge engagement
 
As it way "only" around 30°C max no issues with charging but still  the cooling behavior of the ZOE can be observed. 

Else I find the phase detection feature brillant. Its useful to see with how many phases the car charges. It would be nice to have this included standard in smart charging, but there is one think i don't like as the SmartEVSE alternates the current between 6/8A again ZOE starts/stops the Heat pump for battery/charger cooling. It does not interrupt the charging but its not nice to hear as you can see from the video. I was thinking the charging phases detection algorithm could also use the slow ramp up current to detect how many phases are active without cycling between 6/8A. This could provide more "smooth" charging experience. What do you think?

Car cooling in the background while using Phase detect algorithm:
https://user-images.githubusercontent.com/78184229/184541944-6a53eb86-9eac-4fe3-a3c8-23942fc3e50b.mp4

Car/SmartEVSE while using ramp current startup:
https://user-images.githubusercontent.com/78184229/184542026-91f7c82b-9fa8-4975-9014-b78b140e7d4b.mp4

Car/SmartEVSE while using normal V2.30 charge startup:
https://user-images.githubusercontent.com/78184229/184542125-a479e345-3ac7-4f27-ad6c-e1c740101985.mp4

Will try to leave the car in the sun the whole day and next week as we have some 33°C prediction i hope i can than also make the video of the initial problem. A shame i did not think on recording a video back that when we had this heat wave for two weeks and i had problems every day i was charging...

bobosch, 2022-08-18:
Hi, great thanks for the feedback!

The car have to react to a decreasing current faster than to an increasing current.
To avoid interference with other electricity consumer I only use the decreasing to detect the phase.

In the startup some cars take more than half a minute until they draw the full current, and my car immediately starts charging...

I will try to monitor the current on startup, count the number of phases raising about 6 A and verify with one cycle between 8 A and  6 A

bobosch, 2022-08-19:
[EVSE.zip](https://github.com/SmartEVSE/SmartEVSE-2/files/9381736/EVSE.zip)
This version tries to detect the charging start (0 -> 6 A), and verifies to measurement by increasing from 6 A -> 8 A. Only when this fails, it cycles between 8 A and 6 A. Please give it a try.

AndrejValand, 2022-08-20:
Thanks Robert. Will try this today and report back. 

In the mean time below the videos from two days back when it was 35°C so the CAR started to interrupt the charging, but it was a bit different as when it was 38°C++ for a week and i had the most problems... Now it interrupted the charging at 8A not immediately at 6A, also there was always only one charging restart then it continued to charge normally with all three possible start-up procedures i tried.

What i learned from this is that even ramping up the current slowly does not help in all situations with the ZOE. And there is no way around it that it sometimes disables the charging to start up its cooling at charging start. The main problem i was having where it restarted charging  a few times before it started charging normally, i think now is a timing issue with the ZOE in this special situation in combination with SmartEVSE. SmartEVSE waits quite some time to restart charging. If the car interrupts the charge and in this time the completely shuts down it cooling, when SmartEVSE restarts it started cycling Charge/Restart... Can you please point me to where this time is set in the code i will try to shorten it a bit maybe for 2-5 seconds and i think the problem will be solved. Than it should do only one restart as in vides below in all temperature situations.

Phase detect startup @35°C with one restart:
https://user-images.githubusercontent.com/78184229/185737575-14ba96c9-4456-496e-b820-a33810dde770.MP4

Current Ramp up startup @35°C with one restart:
https://user-images.githubusercontent.com/78184229/185737681-4e5de857-bc45-42c2-be21-8cdc177d3370.mp4

Standard full current start  @35°C with one restart:
https://user-images.githubusercontent.com/78184229/185737701-fefa5858-b872-4177-aead-44ae8e152a0a.mp4

AndrejValand, 2022-08-20:
Was unable to test as its raining like crazy, but did try inside with a simulator. There is one bug if you disconnect charging where DETECTING is on screen it stays on the LCD even when car is disconnected.
[image]

AndrejValand, 2022-08-20:
Rain stopped and i could make some tests with a real car and it has some problems. The current looks ok but it seams to detect false active phases number now.

Try1..3,5:
https://user-images.githubusercontent.com/78184229/185762051-d9f63067-55cd-44f7-bd01-d07cf365cfaa.mp4

Try4:
https://user-images.githubusercontent.com/78184229/185762055-ab074573-b65f-447b-b485-6f7ea0e2637f.mp4

AndrejValand, 2022-08-22:
Was thinking over the weekend. Will phase count detection work ok also in load balancing systems?

bobosch, 2022-08-24:
Untested now. The idea is to pause balancing when measuring.
I fixed the bug, added two constants and changed number of identical measurements from 2 to 3...

#define STARTCURRENT_MEASURE_INCREASE  // Count phases on decreasing and increasing current
#define STARTCURRENT_IDENTICAL_MEASUREMENTS 3 // Number of identical measurements

[EVSE.zip](https://github.com/SmartEVSE/SmartEVSE-2/files/9414792/EVSE.zip)

AndrejValand, 2022-08-24:
The DETECT massage problem is gone. But still i have prolonged reaction time from C->B during the detection phase. This causes the vehicle to enter charger error state. The max reaction for C->B time must be below 100ms this is per EN61851 standard also. This was also my initial problem with the SmartEVSE an the ZOE a few years back, after this reaction time got fixed it worked always. And for unknown reason after a while the power is not updated anymore from the EV meter. It stays at some old value even if charging current changes. The kWh increases normally. Strange.

AndrejValand, 2022-08-24:
Just tested the version from a few week's back (901012e0d9319b2277556e6a8c6c9071016e9b1e  from 19.7.2022) i had downloaded from the repository and there are no problems with C->B detection reaction time. Also power is always displayed correctly when using an additional EV meter. The version had some beginnings for phase detection but probably only for solar. Hope this helps to identify the bug.