SmartEVSE/SmartEVSE-2 #49: XC8 compiler version

lhupfeldt, 2022-11-20:
The documentation [here](http://www.smartevse.nl/building-and-programming) says to use version 1.45 of the XC8 compiler, but the latest (2.40) version seems to work fine now. Should I still use 1.45?

Output from XC8 v2.40 with Ide v6.05:

```
CLEAN SUCCESSFUL (total time: 58ms)
make -f nbproject/Makefile-default.mk SUBPROJECTS= .build-conf
make[1]: Entering directory '/home/lhn/extsrc/SmartEVSE-2/SmartEVSE2.X'
make  -f nbproject/Makefile-default.mk dist/default/production/SmartEVSE2.X.production.hex
make[2]: Entering directory '/home/lhn/extsrc/SmartEVSE-2/SmartEVSE2.X'
"/opt/microchip/xc8/v2.40/bin/xc8-cc"  -mcpu=18F26K22 -c    -fno-short-double -fno-short-float -memi=wordwrite -mrom=0-FCFB -O3 -maddrqual=ignore -xassembler-with-cpp -mwarn=-3 -Wa,-a -DXPRJ_default=default  -msummary=-psect,-class,+mem,-hex,-file  -ginhx32 -Wl,--data-init -mno-keep-startup -mno-download -mdefault-config-bits   -std=c90 -gdwarf-3 -mstack=compiled:auto:auto:auto     -o build/default/production/EVSE.p1 EVSE.c 
"/opt/microchip/xc8/v2.40/bin/xc8-cc"  -mcpu=18F26K22 -c    -fno-short-double -fno-short-float -memi=wordwrite -mrom=0-FCFB -O3 -maddrqual=ignore -xassembler-with-cpp -mwarn=-3 -Wa,-a -DXPRJ_default=default  -msummary=-psect,-class,+mem,-hex,-file  -ginhx32 -Wl,--data-init -mno-keep-startup -mno-download -mdefault-config-bits   -std=c90 -gdwarf-3 -mstack=compiled:auto:auto:auto     -o build/default/production/GLCD.p1 GLCD.c 
"/opt/microchip/xc8/v2.40/bin/xc8-cc"  -mcpu=18F26K22 -c    -fno-short-double -fno-short-float -memi=wordwrite -mrom=0-FCFB -O3 -maddrqual=ignore -xassembler-with-cpp -mwarn=-3 -Wa,-a -DXPRJ_default=default  -msummary=-psect,-class,+mem,-hex,-file  -ginhx32 -Wl,--data-init -mno-keep-startup -mno-download -mdefault-config-bits   -std=c90 -gdwarf-3 -mstack=compiled:auto:auto:auto     -o build/default/production/bootloader.p1 bootloader.c 
"/opt/microchip/xc8/v2.40/bin/xc8-cc"  -mcpu=18F26K22 -c    -fno-short-double -fno-short-float -memi=wordwrite -mrom=0-FCFB -O3 -maddrqual=ignore -xassembler-with-cpp -mwarn=-3 -Wa,-a -DXPRJ_default=default  -msummary=-psect,-class,+mem,-hex,-file  -ginhx32 -Wl,--data-init -mno-keep-startup -mno-download -mdefault-config-bits   -std=c90 -gdwarf-3 -mstack=compiled:auto:auto:auto     -o build/default/production/utils.p1 utils.c 
"/opt/microchip/xc8/v2.40/bin/xc8-cc"  -mcpu=18F26K22 -c    -fno-short-double -fno-short-float -memi=wordwrite -mrom=0-FCFB -O3 -maddrqual=ignore -xassembler-with-cpp -mwarn=-3 -Wa,-a -DXPRJ_default=default  -msummary=-psect,-class,+mem,-hex,-file  -ginhx32 -Wl,--data-init -mno-keep-startup -mno-download -mdefault-config-bits   -std=c90 -gdwarf-3 -mstack=compiled:auto:auto:auto     -o build/default/production/modbus.p1 modbus.c 
"/opt/microchip/xc8/v2.40/bin/xc8-cc"  -mcpu=18F26K22 -c    -fno-short-double -fno-short-float -memi=wordwrite -mrom=0-FCFB -O3 -maddrqual=ignore -xassembler-with-cpp -mwarn=-3 -Wa,-a -DXPRJ_default=default  -msummary=-psect,-class,+mem,-hex,-file  -ginhx32 -Wl,--data-init -mno-keep-startup -mno-download -mdefault-config-bits   -std=c90 -gdwarf-3 -mstack=compiled:auto:auto:auto     -o build/default/production/OneWire.p1 OneWire.c 
"/opt/microchip/xc8/v2.40/bin/xc8-cc"  -mcpu=18F26K22 -c    -fno-short-double -fno-short-float -memi=wordwrite -mrom=0-FCFB -O3 -maddrqual=ignore -xassembler-with-cpp -mwarn=-3 -Wa,-a -DXPRJ_default=default  -msummary=-psect,-class,+mem,-hex,-file  -ginhx32 -Wl,--data-init -mno-keep-startup -mno-download -mdefault-config-bits   -std=c90 -gdwarf-3 -mstack=compiled:auto:auto:auto     -o build/default/production/font.p1 font.c 
"/opt/microchip/xc8/v2.40/bin/xc8-cc"  -mcpu=18F26K22 -c    -fno-short-double -fno-short-float -memi=wordwrite -mrom=0-FCFB -O3 -maddrqual=ignore -xassembler-with-cpp -mwarn=-3 -Wa,-a -DXPRJ_default=default  -msummary=-psect,-class,+mem,-hex,-file  -ginhx32 -Wl,--data-init -mno-keep-startup -mno-download -mdefault-config-bits   -std=c90 -gdwarf-3 -mstack=compiled:auto:auto:auto     -o build/default/production/font2.p1 font2.c 
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
modbus.c:640:44: warning: (373) implicit signed to unsigned conversion
modbus.c:640:129: warning: (373) implicit signed to unsigned conversion
modbus.c:643:40: warning: (373) implicit signed to unsigned conversion
modbus.c:656:27: warning: (373) implicit signed to unsigned conversion
modbus.c:657:31: warning: (373) implicit signed to unsigned conversion
modbus.c:658:27: warning: (373) implicit signed to unsigned conversion
modbus.c:672:15: warning: (373) implicit signed to unsigned conversion
modbus.c:688:99: warning: (373) implicit signed to unsigned conversion
modbus.c:798:36: warning: (373) implicit signed to unsigned conversion
modbus.c:841:60: warning: (373) implicit signed to unsigned conversion
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
EVSE.c:848:22: warning: (346) declaration of "serialnr" hides outer declaration
EVSE.c:856:22: warning: (373) implicit signed to unsigned conversion
EVSE.c:880:47: warning: (373) implicit signed to unsigned conversion
EVSE.c:897:47: warning: (373) implicit signed to unsigned conversion
EVSE.c:920:55: warning: (373) implicit signed to unsigned conversion
EVSE.c:956:44: warning: (373) implicit signed to unsigned conversion
EVSE.c:957:49: warning: (373) implicit signed to unsigned conversion
EVSE.c:959:42: warning: (373) implicit signed to unsigned conversion
EVSE.c:992:49: warning: (373) implicit signed to unsigned conversion
EVSE.c:992:54: warning: (373) implicit signed to unsigned conversion
EVSE.c:993:42: warning: (373) implicit signed to unsigned conversion
EVSE.c:1009:53: warning: (373) implicit signed to unsigned conversion
EVSE.c:1024:64: warning: (373) implicit signed to unsigned conversion
EVSE.c:1026:67: warning: (373) implicit signed to unsigned conversion
EVSE.c:1027:1: warning: (373) implicit signed to unsigned conversion
EVSE.c:1027:59: warning: (373) implicit signed to unsigned conversion
EVSE.c:1029:42: warning: (373) implicit signed to unsigned conversion
EVSE.c:1085:78: warning: (373) implicit signed to unsigned conversion
EVSE.c:1103:41: warning: (373) implicit signed to unsigned conversion
EVSE.c:1187:66: warning: (373) implicit signed to unsigned conversion
EVSE.c:1188:35: warning: (373) implicit signed to unsigned conversion
EVSE.c:1285:37: warning: (373) implicit signed to unsigned conversion
EVSE.c:1605:28: warning: (373) implicit signed to unsigned conversion
EVSE.c:1609:28: warning: (373) implicit signed to unsigned conversion
EVSE.c:1613:28: warning: (373) implicit signed to unsigned conversion
EVSE.c:1617:28: warning: (373) implicit signed to unsigned conversion
EVSE.c:1804:22: warning: (373) implicit signed to unsigned conversion
EVSE.c:2188:50: warning: (373) implicit signed to unsigned conversion
EVSE.c:2272:43: warning: (373) implicit signed to unsigned conversion
EVSE.c:2339:25: warning: (373) implicit signed to unsigned conversion
EVSE.c:2340:23: warning: (373) implicit signed to unsigned conversion
EVSE.c:2341:33: warning: (373) implicit signed to unsigned conversion
EVSE.c:2342:30: warning: (373) implicit signed to unsigned conversion
EVSE.c:2352:20: warning: (373) implicit signed to unsigned conversion
EVSE.c:2367:22: warning: (373) implicit signed to unsigned conversion
EVSE.c:3087:86: warning: (373) implicit signed to unsigned conversion
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
utils.c:39:28: warning: (373) implicit signed to unsigned conversion
utils.c:105:130: warning: (373) implicit signed to unsigned conversion
utils.c:139:28: warning: (373) implicit signed to unsigned conversion
[... 18 lines cut ...]
GLCD.c:274:34: warning: (373) implicit signed to unsigned conversion
GLCD.c:288:58: warning: (373) implicit signed to unsigned conversion
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
[... cut ...]
    EEPROM space         used     0h (     0) of   400h bytes   (  0.0%)
    ID Location space    used     8h (     8) of     8h bytes   (100.0%)

make[2]: Leaving directory '/home/lhn/extsrc/SmartEVSE-2/SmartEVSE2.X'
make[1]: Leaving directory '/home/lhn/extsrc/SmartEVSE-2/SmartEVSE2.X'

BUILD SUCCESSFUL (total time: 17s)
Loading code from /home/lhn/extsrc/SmartEVSE-2/SmartEVSE2.X/dist/default/production/SmartEVSE2.X.production.hex...
Program loaded with pack,PIC18F-K_DFP,1.7.134,Microchip
Loading completed
```

lhupfeldt, 2022-11-20:
Should I worry about `::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2`?

mstegen, 2022-11-21:
2.40 should be fine. I'll update that website soon.

bobosch, 2022-11-22:
> Should I worry about

No, the free version of the compiler produces less optimized code which consumes more memory. As long as it fit into RAM... I use the free version, too.