SmartEVSE/SmartEVSE-2 #50: Loading hex file from Linux

lhupfeldt, 2023-04-22:
I'm on Linux.
I have compiled the project and now I need to load it.

The instructions here https://www.smartevse.nl/building-and-programming say "Start the AN1310 bootloader program in the /bootloader directory."
I cannot find any program  called AN1310, and no bootloader directories related to mplab.

Should I use UnifiedHost-1.19.1 booloader from here https://www.microchip.com/en-us/tools-resources/develop/libraries/microchip-bootloaders/8-bit?
If so, then what are the correct settings?

Or something completely different?

bobosch, 2023-04-22:
Hi, I use Linux, too.
Good questions, some years ago it was available. It runs fine on wine :-)
Regarding this document https://ww1.microchip.com/downloads/en/AppNotes/01310a.pdf
[image]
even the source code was free to download, also available here https://github.com/martonmiklos/AN1310-Qt5
So I think it is OK to share the original archive with the windows executable here
[Serial Bootloader AN1310 v1.05r.zip](https://github.com/SmartEVSE/SmartEVSE-2/files/11301779/Serial.Bootloader.AN1310.v1.05r.zip)

lhupfeldt, 2023-04-22:
Hi @bobosch,

Thank you for your quick reply. I would rather not have to install wine just for this if there is a simpler solution. 

UnifiedHosts did not work with java 17, I changed to 1.8 and then it started

`sudo alternatives --config java`

select `java-1.8...`

`...UnifiedHost-1.19.1]$ java -Djava.util.logging.config.file=logging.properties -jar UnifiedHost-1.19.1.jar `

[image]

I have not tried using it, as I don't know which options to select.

mstegen, 2023-04-25:
The Unified Bootloader is unfortunately not compatible with the AN1310 bootloader, so this will not work.
I think the only option right now is to compile the sourcefiles somehow, or run wine.

The windows bootloader can be found here:
https://github.com/SmartEVSE/smartevse/tree/master/bootloader

lhupfeldt, 2023-05-21:
Thank you both for your replies.
I finally got around to this again, and would like to provide some information if other people that may attempt loading firmware from Linux.

I could not make the AN1310 application running under `wine` recognise the usb serial port. It shows up correctly in linux as `/dev/ttyUSB0` when the cable is plugged into the laptop, and it shows up automatically in `.wine/dosdevices` as com33 and com34. I also tried symlinking it to com1. I have `rw` permissions to the device (I added myself to the `dialout` group). However the AN1310 is not showing the serial USB device.

I have instead compiled the program for Qt5 from the sources mentioned above, and the usb device is showing up (I have not yet attempted to load the hex file onto SmartEVSE).
The compilation was rather complicated, the instructions in the Readme.txt were insufficient.

Follow steps 1-3 from the Readme.txt (or replace step 1 with installation of `QT Creator` using your package manager)
Before step 4 (compiling  AN1310ui/AN1310ui.pro) I had to add `Bootload` and `QextSerialPort` as external libraries, as described here https://doc.qt.io/qtcreator/creator-project-qmake-libraries.html. Add the archive files (.a) from the build-Bootload/QextSerialPort-Desktop-Release and add the include dir as `Bootload/QextSerialPort` respectively.

lhupfeldt, 2023-05-21:
[image]

bobosch, 2023-05-22:
With wine you have to add a registry key (wine regedit) to
Path: HKEY_LOCAL_MACHINE\Software\Wine\Ports
Name: COM1
Type: REG_SZ
Value: /dev/ttyUSB0

lhupfeldt, 2023-05-22:
I actually tried that as well, forgot to mention that. Unfortunately I already uninstalled wine after building the native Linux AN1310 application, so I can't double check that I entered exactly what you suggest.

lhupfeldt, 2023-06-25:
For anybody coming here:
The https://github.com/SmartEVSE/SmartEVSE-2/files/11301779/Serial.Bootloader.AN1310.v1.05r.zip does NOT work. It goes into an endless flashing loop, as described here https://github.com/martonmiklos/AN1310-Qt5/issues/2. This https://github.com/MrFillbert/AN1310-v1.05-QSerialPort works.