SmartEVSE/SmartEVSE-2 #52: Cannot open include file error

lhupfeldt, 2023-05-21:
I'm attempting to build the project using MPLAB X IDE 6.10, but I'm getting the followng error:

```

EVSE.c:3103:86: warning: (373) implicit signed to unsigned conversion
"/opt/microchip/xc8/v2.41/bin/xc8-cc"  -mcpu=18F26K22 -Wl,-Map=dist/default/production/SmartEVSE2.X.production.map  -DXPRJ_default=default  -Wl,--defsym=__MPLAB_BUILD=1   -mdfp="/home/lhn/.mchp_packs/Microchip/PIC18F-K_DFP/1.1.58/xc8"  -fno-short-double -fno-short-float -memi=wordwrite -mrom=0-FCFB -O3 -maddrqual=ignore -xassembler-with-cpp -mwarn=-3 -Wa,-a -msummary=-psect,-class,+mem,-hex,-file  -ginhx32 -Wl,--data-init -mno-keep-startup -mno-download -mdefault-config-bits -std=c90 -gdwarf-3 -mstack=compiled:auto:auto:auto      -Wl,--memorysummary,dist/default/production/memoryfile.xml -o dist/default/production/SmartEVSE2.X.production.elf  build/default/production/EVSE.p1 build/default/production/GLCD.p1 build/default/production/bootloader.p1 build/default/production/utils.p1 build/default/production/modbus.p1 build/default/production/OneWire.p1 build/default/production/font2.p1 build/default/production/font.p1     
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
Non line specific message::: advisory: (1492) using updated 32-bit floating-point libraries; improved accuracy might increase code size
modbus.c:765:: warning: (765) degenerate unsigned comparison
/tmp/xcXLCGDGG/driver_tmp_1.s:8:: error: (140) can't open include file "/home/lhn/.mchp_packs/Microchip/PIC18F-K_DFP/1.1.58/xc8/pic/include/proc/18f26k22.cgen.inc": No such file or directory
/tmp/xcXLCGDGG/driver_tmp_1.s:29758:: error: (800) undefined symbol "tblptrl"
/tmp/xcXLCGDGG/driver_tmp_1.s:29760:: error: (800) undefined symbol "tblptrh"
...
```

The pack is installed:
```
 ls -d .mchp_packs/Microchip/PIC18F-K_DFP/1.1.58/xc8/pic/include/
.mchp_packs/Microchip/PIC18F-K_DFP/1.1.58/xc8/pic/include/
```
but there is no `proc` in the `include` dir

I have tried reinstalling the pack, but still no `proc`.

I successfully built this earlier using 6.05

lhupfeldt, 2023-05-21:
Forgot to mention that I have also done a `git pull` since last successful build.

mstegen, 2023-05-23:
I'm sorry, but i'm unable to help you on this.
I'm using MPLABX on WIN10. 
Using the same PIC18F-K_DFP version 1.1.58

If it's only this error, you might comment out the "read serialnr" function where tblptr is used.

bobosch, 2023-05-24:
I updated to MPLABX IDE 6.10 and XC8 compiler 2.41 - works fine with linux here.

bobosch, 2023-05-24:
Seems that there are some lines missing on your snipset
```
EVSE.c:3103:86: warning: (373) implicit signed to unsigned conversion
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
OneWire.c:184:43: warning: (373) implicit signed to unsigned conversion
OneWire.c:205:40: warning: (373) implicit signed to unsigned conversion
OneWire.c:230:44: warning: (373) implicit signed to unsigned conversion
OneWire.c:263:22: warning: (373) implicit signed to unsigned conversion
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
GLCD.c:92:28: warning: (373) implicit signed to unsigned conversion
GLCD.c:99:35: warning: (373) implicit signed to unsigned conversion
GLCD.c:101:30: warning: (373) implicit signed to unsigned conversion
GLCD.c:138:22: warning: (373) implicit signed to unsigned conversion
GLCD.c:161:22: warning: (373) implicit signed to unsigned conversion
GLCD.c:165:18: warning: (373) implicit signed to unsigned conversion
GLCD.c:173:31: warning: (373) implicit signed to unsigned conversion
GLCD.c:177:18: warning: (373) implicit signed to unsigned conversion
GLCD.c:198:21: warning: (373) implicit signed to unsigned conversion
GLCD.c:201:29: warning: (373) implicit signed to unsigned conversion
GLCD.c:222:38: warning: (373) implicit signed to unsigned conversion
GLCD.c:271:40: warning: (373) implicit signed to unsigned conversion
GLCD.c:274:34: warning: (373) implicit signed to unsigned conversion
GLCD.c:288:58: warning: (373) implicit signed to unsigned conversion
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
utils.c:39:28: warning: (373) implicit signed to unsigned conversion
utils.c:105:130: warning: (373) implicit signed to unsigned conversion
utils.c:139:28: warning: (373) implicit signed to unsigned conversion
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
modbus.c:648:44: warning: (373) implicit signed to unsigned conversion
modbus.c:648:129: warning: (373) implicit signed to unsigned conversion
modbus.c:651:40: warning: (373) implicit signed to unsigned conversion
modbus.c:664:27: warning: (373) implicit signed to unsigned conversion
modbus.c:665:31: warning: (373) implicit signed to unsigned conversion
modbus.c:666:27: warning: (373) implicit signed to unsigned conversion
modbus.c:680:15: warning: (373) implicit signed to unsigned conversion
modbus.c:696:99: warning: (373) implicit signed to unsigned conversion
modbus.c:806:36: warning: (373) implicit signed to unsigned conversion
modbus.c:849:60: warning: (373) implicit signed to unsigned conversion
"/opt/microchip/xc8/v2.41/bin/xc8-cc"  -mcpu=18F26K22 -Wl,-Map=dist/default/production/SmartEVSE2.X.production.map  -DXPRJ_default=default  -Wl,--defsym=__MPLAB_BUILD=1   -mdfp="/home/robert/.mchp_packs/Microchip/PIC18F-K_DFP/1.1.58/xc8"  -fno-short-double -fno-short-float -memi=wordwrite -mrom=0-FCFB -O3 -maddrqual=ignore -xassembler-with-cpp -mwarn=-3 -Wa,-a -msummary=-psect,-class,+mem,-hex,-file  -ginhx32 -Wl,--data-init -mno-keep-startup -mno-download -mdefault-config-bits -std=c90 -gdwarf-3 -mstack=compiled:auto:auto:auto      -Wl,--memorysummary,dist/default/production/memoryfile.xml -o dist/default/production/SmartEVSE2.X.production.elf  build/default/production/EVSE.p1 build/default/production/GLCD.p1 build/default/production/bootloader.p1 build/default/production/utils.p1 build/default/production/modbus.p1 build/default/production/OneWire.p1 build/default/production/font2.p1 build/default/production/font.p1     
::: advisory: (2051) the current license does not permit the selected optimization level, using optimization level 2
Non line specific message::: advisory: (1492) using updated 32-bit floating-point libraries; improved accuracy might increase code size
modbus.c:767:: warning: (765) degenerate unsigned comparison

bobosch, 2023-05-25:
This is version 2.30, here is the latest:
[SmartEVSE2.X.production.hex.zip](https://github.com/SmartEVSE/SmartEVSE-2/files/11568977/SmartEVSE2.X.production.hex.zip)

bobosch, 2023-05-27:
@lhupfeldt Does the phase count detection (new feature in latest version) works on your setup? I've changed the balancing algorithm in solar mode depending on the charging phases here: 
[SmartEVSE2.X.beta.hex.zip](https://github.com/SmartEVSE/SmartEVSE-2/files/11581267/SmartEVSE2.X.beta.hex.zip)

lhupfeldt, 2023-05-29:
@bobosch Sorry, I have not been at home for the last couple of days, so I did not attempt to load the hex file yet. Would you like me to load the [SmartEVSE2.X.production.hex.zip](https://github.com/SmartEVSE/SmartEVSE-2/files/11568977/SmartEVSE2.X.production.hex.zip)  to test that? Or [SmartEVSE2.X.beta.hex.zip](https://github.com/SmartEVSE/SmartEVSE-2/files/11581267/SmartEVSE2.X.beta.hex.zip) ? I can try to load either if you want me to test.

bobosch, 2023-05-31:
Please test the beta.hex - if you are facing issues on balancing you can fall back to the production.hex

lhupfeldt, 2023-06-25:
Hi @bobosch I finally got around to trying the beta.hex.
Now I just need to know what you want me to test :)
Where is the phase count supposed to show up?
The main display, or in a menu?

https://github.com/SmartEVSE/SmartEVSE-2/assets/1584329/41dcc5ac-87da-4d97-b692-327b131a33cb

lhupfeldt, 2023-06-25:
BTW, the small fonts on the main display are hard to read outside through the enclosure door plastic. It would be great with a slowly toggling display with large fonts.

bobosch, 2023-06-26:
> Now I just need to know what you want me to test :)

The automatic phase detection and the balancing (keep the mains current at 0 A while solar charging)

bobosch, 2023-06-26:
> BTW, the small fonts on the main display are hard to read outside through the enclosure door plastic

Even when the backlight is on?

lhupfeldt, 2023-06-26:
> > Now I just need to know what you want me to test :)
> 
> The automatic phase detection and the balancing (keep the mains current at 0 A while solar charging)

You want all three phases to show 0 on the SmartEVSE main display while solar charging?
I'm afraid I may need a more specific description of what you want me to look for :)

bobosch, 2023-06-27:
> It is normally off and it would be nice to be able read the display when the enclosure door is closed.

I have also a enclosure, but when the backlight is on the small font is readable. When the backlight is off even the big font is not good readable.
While charging most of the information is shown with the big font - what do you miss?