SmartEVSE/SmartEVSE-2 #60: Old sensorbox comm's died when updated SmartEVSE v2 to 2.3

juntta, 2024-11-08:
I’ve got two SmartEVSE v2 in smart mode which I have to upgrade v2 to newer version to support external switch. 

After I updated both v2’s I’ve been getting ”no serial com, check wiring” errors. I double checked config menu and everything should be in order.

Isn’t the old sensorbox (v1.1) compatible any more with 2.20 and newer versions? Is the sensorbox possible to update with ftdi cable (or some other means)?

mstegen, 2024-11-08:
If you use an old sensorbox 1.0 or 1.1 then it's not compatible anymore with the latest smartEVSE v2 firmware.
The latest versions use modbus, while the old version used a different (slower) protocol.

You can flash the sensorbox 1 to a newer firmware (available in the sensorbox 1 repo), but you will need to use the pickit or other microchip programmer to program it.

juntta, 2024-11-08:
I got the sensorbox updated with Pickit programmer and now smartmode on SmartEVSE v2 functions normally.

Thanks @mstegen

closed 2024-11-08 by juntta